luc-github/ESP3D #1038: [BUG] cannot get the esp32 and the board to communicate

Reza-v, 2024-07-26:
**Describe the bug**

the esp32 and the 3d printer mainboard do not want to communicate over serial, no matter the pins or uart i use. i am using a lolin esp32 s2 mini and a mega 2560 based board, using the rx0 and tx0 bootloader serial pins.

**To Reproduce**

1. uploaded code to esp32
2. connected the esp32 s2 to the mainboard using jumpers  with a voltage divider on the tx pin coming from the mega (have confirmed the rx and tx pins used on the mainboard do work as intended and the problem is with the esp

i have tried using the default pins in pins.h as well as selecting another set of pins.
**Screenshots**

[image]

**ESP3D Firmware:**

- ESP3D FW version: [3.0] 
- ESP3D-WebUI Version:[3.0]
- Wifi mode:[STA]
- Flash method: [PlatformIO]  
- Any change done in FW: only the changes in configuration.h

**Target Firmware:**

- Name: [Marlin]
- Version [2.1.2.4]

**Board used (please complete the following information):**

- MCU:[esp32 s2}
- Name:[lolin s2 mini]
- Flash size: [4M]

luc-github, 2024-07-27:
that screen is not related to serial communication but communication between webUI and esp3D
what version of webUI are you using ?
what version of esp3d are you using ?
type `http:\\<your-IP>?forcefallback=yes` to go to maintenance page and delete the current webUI which is either bad version or corrupted

Reza-v, 2024-07-27:
I clicked on "latest development version" on the github page and downloaded that, built it and uploaded it. i will try doing it again today. thank you for the help.

luc-github, 2024-07-27:
webui packages are already build in dist directory, what about esp3d version , it is visible in maintenance page

Reza-v, 2024-07-27:
[image]
this is the one

luc-github, 2024-07-27:
ok this is latest version

Reza-v, 2024-07-27:
Ok so the prebuilt webui worked like a charm and I'm past the connecting page but now I still can't communicate with the board, the gcodes I send are given no response. I tried reuploading the code to the ESP, using default pins, using serial0 and 1, both with default rx tx pins and different pins, nothing is working.

i have used pins 4 and 5 as the default rx tx pins for serial1 suggested by you in another issue thread, no dice,

also I must point out that sometimes when I connect the jumpers, the wifi just dies and I need to reset the board to get it working again. is that something you've seen before?

luc-github, 2024-07-27:
connect the jumpers ? what jumpers are you talking about?

Reza-v, 2024-07-27:
the jumpers to connect tx and rx pins to the 3d printer board.
i am using a voltage divider on the tx pin coming from the 2560

luc-github, 2024-07-27:
you hàve a picture of your wiring ? I do not understand what you mean

Reza-v, 2024-07-27:
what's so hard to understand? i just don't know what pins on the esp32 s2 is the default serial1 pins and no one seems to know either, and nothing i try works. shouldn't you connect the esp board to the 3d printer board through serial in order for them to communicate with each other? that's my main issue. i cannot get them to communicate with each other no matter what pins i try or what pins i set the esp32 to recognize as tx and rx.

luc-github, 2024-07-27:
looking at https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/_images/esp32-s2_saola1-pinout.jpg
TX1 = GPIO17 / RX1 = GPIO18  is that what you use for serial 1 or do you use custom pins ?
https://esp3d.io/esp3d/v3.x/hardware/esp_boards/esp32-s2/wemos-s2-mini/index.html

Reza-v, 2024-07-27:
oh you must be confused, i am talking about jumper wires not the other type of jumpers. i am using jumper wires temporarily for testing and i'll make proper connectors after it actually works.

unfortunately pins 17 and 18 did not work.

luc-github, 2024-07-27:
then in configuration.h
use Serial0
`#define ESP_SERIAL_OUTPUT USE_SERIAL_0`

and add these 2 lines
```
#define ESP_RX_PIN 37
#define ESP_TX_PIN 39
```

Reza-v, 2024-07-27:
what about the lines in esp3d_pins.h file? those don't matter? i've been changing those values to custom pins

luc-github, 2024-07-27:
in all case I suggest you to use this kind of device : https://www.amazon.com/HiLetgo-FT232RL-Converter-Adapter-Breakout/dp/B00IJXZQ7C/ref=sr_1_2_sspa

to check your esp board pins and also to check your mega wiring is correct - you will save time in debug
be sure baudrate is same on both side

Reza-v, 2024-07-27:
how do you change baud rate in your code? i could not find it

luc-github, 2024-07-27:
in when UI not in code

Reza-v, 2024-07-27:
oh that's interesting...

luc-github, 2024-07-27:
or using direct commands : https://esp3d.io/esp3d/v3.x/documentation/commands/esp901/index.html

Reza-v, 2024-07-27:
that worked. sorry i wanted to ask you how to change the baud rate but i totally forgot. i never got to the UI and just assumed it's stuck on connecting because i didn't get the serial hooked up correctly. thanks a lot for the help.

closed 2024-07-27 by Reza-v