luc-github/ESP3D #1039: [FEATURE REQUEST]programming the 3d printer mainboard using wifi

Reza-v, 2024-07-28:
I'm not sure if you can already do this, i tried without success but the idea is to use the esp32 as a serial bridge over wifi and upload firmware over wifi through uart, just like you would with an hc05 or similar serial bridges. I tried using the ip address as a com port in platformio, it used to work on esp3dlib on the mks tinybee, same with relatively ok upload speed to the sd card but i gather that has been removed in the esp3d due to slow speeds. I also saw some pin specifications in the config file for the esp3d for an sd card. Can you run gcode from the sd card on the esp32 or does it have to be in the sd card on the printer mainboard? I assume you can actually upload files to th3 sd card connected to the esp32 so if you could also print from it that would be great and if you can't, what is the purpose of it?

luc-github, 2024-07-28:
programming any printer board via an esp board is not easy due to different way boards can be flashed and because this not something you do often there is no real added value to spend time on this.

the gcode host streaming is still under development : https://github.com/luc-github/ESP3D/issues/243 

usage of SD / connected or not is described here : http://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html

closed 2024-07-28 by luc-github

Reza-v, 2024-07-28:
Can't you yoink some of the code from marlin developed for esp32 and use it so that you don't have to start from nothing? Also when the host streaming is available, do you retain settings like acceleration, pressure advance and input shaping from marlin or would that be calculated by the host like octoprint does?

luc-github, 2024-07-28:
>Can't you yoink some of the code from marlin developed for esp32

what code? what do you mean?

>do you retain settings like acceleration, pressure advance and input shaping from marlin

no : streaming gcode means streaming gcode, what you list is not related to streaming and is managed by Marlin