luc-github/ESP3D #1040: [FEATURE REQUEST]Add ESP32-Cam board to the 2.1.x

amirzohaib, 2024-07-31:
**Is your feature request related to a problem? Please describe.**
Using Platform IO, there are only a few options, ESP32 dev, ESP8266..etc but there should be esp32cam
Just like in Beta 3.0x

Why not use the new 3? well you removed the only feature worth going through the trouble of adding WIFI to an old Printer = send STL files via WIFI to the printer.

i know, it was slow, but i am lazy and willing to let it take its time uploading STLs wirelessly. instead of running around pulling SD cards.

**Describe the solution you'd like**

Either add esp32cam to old 2.1x version, or add ability to send stl files via Wifi in 3.0x

**Describe alternatives you've considered**

hardwiring to exisitng SD card on printer

**Additional context**

configure.h comments are sometimes not undesrstandable and confusing

luc-github, 2024-07-31:
please read https://esp3d.io I am sure you will find some information about SD card support in 3.0 it is not because M28/M29 protocol is dropped there is no SD Card support 

Code of 3.0 is totaly different than 2.1X so backporting a feature will be painful but you can do if you want 
I have no time for that sorry

closed 2024-07-31 by luc-github

amirzohaib, 2024-07-31:
Its okay we are all Lazy haha

> please read https://esp3d.io I am sure you will find some information about SD card support in 3.0 it is not because M28/M29 protocol is dropped there is no SD Card support and because configuration.h is now complex there is a configurator for it.
> 

i have used the configurator and enabled the SD card..etc but
there is no upload button on the 3.0xUI

luc-github, 2024-07-31:
what SD card do expect to use for upload ?
The  supported ones :
https://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html