luc-github/ESP3D #1045: [BUG]<Enter comprehensive title>

regDi, 2024-08-27:
**no comunication ESP32cam and Anet A8**

<I followed the steps described on the "https://esp3d.io/ESP3D/Version_3.X/installation/arduino/index.html" for uploading the code to the ESP32 and the modification of the board of the Anet A8, but whem i conect the ESP to the board i have no response at all the of the printer. The conexion to the WIFI is OK, but i can not comunicate. 

[image]

**To Reproduce**

About the instalatin i followed the step described, and i also verified the schematic of the boards to make sure it is well conected. The only responce i get from the printer is M503.

[image]

**Expected behavior**

**Screenshots**

**ESP3D Firmware:**

- ESP3D FW version3
- ESP3D-WebUI Version3
- Wifi mode:[e.g. STA/AP]
- Flash method: [Arduino Ide     If you build with Arduino:
    - Arduino ESP32 cam
- Any change done in FW:

**Target Firmware:**

Printer ANET A8 - v1.5
- Name: Marlin
- Version 1.1.9

**Board used (please complete the following information):**

- MCU:[ESP32]
- Name: ESP32 Dev Module
- Flash size: 4MB

I hope with this information you can help me. thanks a lot in advance.

luc-github, 2024-08-27:
I do no t see any response from M503 form your screen shot

To test use M105 and M114 and M115 

Here list of things you need to check :
0 - Based on https://esp3d.io/ESP3D/Version_3.X/hardware/system_boards/main_boards/anet_a8/ the serial is same as USB, so do all test without USB connected 
1 - baud rate, be sure is same on both side (Marlin and ESP3D)
2 - the voltage dividor, because board is 5V and esp32 is 3.3.v, you must have one on RX at least
3 - Check your wiring, common mistake is to invert RX and TX, or forget to share the ground 
4 - because I am guessing you used some usb 2 serial board to flash the ESP32Cam, use it to check the communication :
          a  - with your esp32cam but not in flash mode, open arduino serial monitor an see if your board communicate, which should be ok as you where able to flash it using serial pins
          b - use RX/TX /GND/5V on the board connect to the usbserial adpter , use Arduino Serial Monitor to check that you can communicate with your board/Marlin

regDi, 2024-08-28:
Finally the issue might be in the print board. I have no communication, thanks a lot for the quick response and for share the proyect. 
I am thinking in change the board of the printer, do you have a recommendation?

luc-github, 2024-08-29:
check your soldering is ok - other people succeded , the  https://esp3d.io/ESP3D/Version_3.X/hardware/system_boards/main_boards/anet_a8/ is a report from a a8 owner

about others boards, It depends on the budget and the desired features, the firmware you want, the possibilities for upgrades, the choice is vast

regDi, 2024-08-29:
Yes that was the issue, the soldering. I am looking for a place to see if they can repair it. I will tell you if i am finally able to finish the project. thank you!

luc-github, 2024-08-29:
Ok good luck , I close ticket  then

closed 2024-08-29 by luc-github