luc-github/ESP3D #1051: Feature request. Adjustable delay for file transfer.

A13xund3r, 2024-09-18:
### Discussed in https://github.com/luc-github/ESP3D/discussions/1010

<div type='discussions-op-text'>

<sup>Originally posted by **A13xund3r** March 19, 2024</sup>
I was repairing BTT GTR V1 mainboard at work and took opportunity to check how does ESP3D communicate during file upload. So this is scope screenshot, yellow line is data sent by ESP3D, blue line is GTR response.
[image]
As you can see GTR respond literally immediately after ESP finish line, I didn't sniffed what it is but I'm 100% sure it is "OK". So why ESP wait 20ms before it send another line? I suspect it is safety measure to give mainboard time to process data it just received but I expect it is already processed when it reply "OK". Also BTT GTR is a monster comparing to any Atmel based controller so I think we can play with shorter or no delay at all. At least it could be available as adjustable option in settings.

</div>

luc-github, 2024-09-18:
What file transfert are you refering to? And what version of ESP3D are you using?

A13xund3r, 2024-09-18:
Old style M28/M29 file transfer protocol, ESP3D2.1.x

luc-github, 2024-09-18:
This protocol is no more supported https://esp3d.io/ESP3D/Version_2.1.X/documentation/sdtransfert/

A13xund3r, 2024-09-18:
Like big corpo discontinuing functional products... 
Could you at least point me to the file with command holding CPU for 20ms during file transfer?

luc-github, 2024-09-18:
I am not Big corpo, I am not paid for what I do
I share my code for free, so such comment is pretty rude
If you are not happy with ESP3D do not use it, that is very simple, my code is fully available so do your homework by yourself, I am not willing to give any help after such attitude

A13xund3r, 2024-09-18:
Won't fix

closed 2024-09-18 by A13xund3r

A13xund3r, 2024-09-18:
Absolutely not bad day, you just misinterpreted my comment which was meant to have humoristic attitude. Especially when my work does exist mainly because big fishes in industrial business say they no longer support something and offer expensive upgrade so we are here to fix things.
So as you see both of us doesn't see anything wrong in our words but does see in each other. Normal in txt communication where body language is missing.