luc-github/ESP3D #1053: [BUG] SD stream failed: 1

piotr-go, 2024-10-16:
**Describe the bug**
**To Reproduce**

Sending a "[ESP700]/SD/test.gcode" in web gui terminal gives me:
> [ESP700]/SD/test.gcode
> ok
> error: stream failed: 1

**Expected behavior**
no error

**ESP3D Firmware:**

ESP3D v3

**Target Firmware:**

Marlin 2.1.x

**Board used (please complete the following information):**

ESP32-S3 (4MB flash, 2MB PSRAM)
SKR-Mini E3 v1.2

**Additional context**

I checked what serial output looks like, and there is no '\n' after command.
If i fake "ok" answer with serial terminal (ESP32---uart---PC ), i sill got "error: stream failed: 1".
It seems there is no reaction to incoming data.

Two way terminal communication in web gui is working. Two way telnet-uart also OK.

piotr-go, 2024-10-16:
1. I don't understand, I thought that communication with Marlin is basic ping/pong streaming?
2. I uploaded this file over wifi. Missing file is error 12, error 1 is answer timeout.

luc-github, 2024-10-16:
1 not really webui send commands and analyse serial from Marlin
2 you upload to which sd ? your above description does not mention any

luc-github, 2024-10-16:
you should read this:
https://esp3d.io/ESP3D/Version_3.X/documentation/sdtransfer/

piotr-go, 2024-10-16:
2. to ESP_DIRECT_SD, file is there, i confirmed. MD5 sum mach, no errors.

piotr-go, 2024-10-16:
Yes.
So no SD streaming in v3 for now?

luc-github, 2024-10-16:
not in v3 yet,  streaming properly is not that easy
ESP3D-TFT has basic Marlin ping pong streaming but it is slow on small movement,  must add buffered streaming to be good.
so still some work to do
will be same.for V3

piotr-go, 2024-10-16:
Is it possible to run v2 on esp32-s3 (or s2) ? Is SD streaming supported in v2?

luc-github, 2024-10-16:
sd streaming is not supported in v2, direct sd upload is even not supported in v2

closed 2024-10-16 by piotr-go

luc-github, 2024-10-16:
have look to https://esp3d.io/ there are a lot of information