luc-github/ESP3D #1054: [Question] How to print over WiFi from Ultimaker Cura 4

piotr-go, 2024-10-16:
I have:
- Ender 3 Pro printer with SKR mini E3 v1.2 board (Marlin 2.1.x)
- ESP32-S3 (or S2) module with ESP3D v3 FW, connected over uart to ender, two way communication is working

How to set Ultimaker Cura 4 (best 3d print results i have) in linux to print over WiFi?

luc-github, 2024-10-17:
Streaming over WiFi can be challenging due to network latency even with strong network.
About cura, I think it has a pluggin for network connection but I do not remember name sorry

piotr-go, 2024-10-17:
Couldn't the transfer be buffered? 2MB is plenty.

As an experiment, I started writing my own print server.

luc-github, 2024-10-18:
> Couldn't the transfer be buffered? 2MB is plenty.
> 
I am not sure to understand , buffering should be either in Printer Firmware, either in streaminng host that do buffered streaming
ESP3D is just a bridge it can not do buffering unless it become itself a streaming host

> As an experiment, I started writing my own print server.

Good that is nice project - you will see that is no so easy to do proper gcode streaming, it is really a full project on its own actually

piotr-go, 2024-10-18:
> buffering should be either in Printer Firmware

That would be the best, but the printer buffer is small.

> Good that is nice project - you will see that is no so easy to do proper gcode streaming, it is really a full project on its own actually

Printing is already partially working. It doesn't seem to be slower than printing from SD (I'll measure the time when I fix everything). Unfortunately, printing stops halfway through. I need to find a bug in the code.

luc-github, 2024-10-18:
in ESP3D-TFT the slowest part is the circle, many small movement, other scommands are ok for the speed I think

Does your sender do also Temperature / Position monitoring   when streaming ?

Also does it handle heating / busy / processing messages that block all incoming commands ? as well as out of filament message ?

What language do you use ?

Sorry many questions but I am curious 😅

piotr-go, 2024-10-18:
> Does your sender do also Temperature / Position monitoring when streaming ?

Sends all responses to PC, but they are ignored for now.

> Also does it handle heating / busy / processing messages that block all incoming commands ?

I calculate the printer buffer level based on "ok" responses and try to fill it to the maximum.

> as well as out of filament message ?

Errors and unknown commands not yet supported. Work in progress.

> What language do you use ?

C

I'm currently working on printing from a file.
Web gui file upload based on esp-idf example already working.

piotr-go, 2024-10-20:
Success.
I printed from SD card connected to ESP32-S3.

~32MB file printing time:
- from printer card: 6h25m +-1m.
- from ESP32 card: 6h25m +-1m.

No difference in print quality.

[image]
File: a sample from Ender 3 Pro card.
Filament: 5 year old white (no idea what type, no label), also sample.

ESP32 to printer transfer baud rate: 115200bps.
WiFi download speed: ~410kB (sorry, upload not measured).
Aaaaaa, card connected over SPI.

luc-github, 2024-10-21:
do you plan to share your code ?

piotr-go, 2024-10-21:
Probably yes. But it will take couple weeks.
I designed and ordered new PCB. For the prototype I used my internet radio ESP32-S3-MINI PCB.
Final version (i hope :)) will be based on ESP32-C3-WROOM (less pain in the ass to solder).

Prototype:
[image]
C3-WROOM:
[image]

USB for programming, µSD, 2*UART, I2C, power connector, button, LED.

piotr-go, 2024-10-21:
Final version will probably have only wifi upload and stream from card. At least that is the goal of this project.

luc-github, 2024-10-21:
so not streaming progression information neither printer status monitoring ?

piotr-go, 2024-10-21:
Print progress is available on LCD and will be available in web gui.

luc-github, 2024-10-21:
> Print progress is available on LCD and will be available in web gui.

Sorry I am confused because you wrote:

,>Final version will probably have only wifi upload and stream from card. At least that is the goal of this project.

fyi just loading webui may bring delay on mono core

piotr-go, 2024-10-21:
I can set high priority for the print thread.
I am happy with just sending file over wifi + starting printing. This is already big improvement.
If wifi print status does not work on a single core, I will not be hugely disappointed.

luc-github, 2024-10-21:
Ok - I close issue then

closed 2024-10-21 by luc-github