luc-github/ESP3D #1056: [FEATURE REQUEST]Add ADXL345 accelerometer for input shaping

ilbossdelfaidate, 2024-10-20:
If it is possible to add an adxl345 accelerometer (even just among the plugins) to calculate the resonant frequencies of the printer to be inserted in the input shaping parameter

luc-github, 2024-10-21:
this is for Klipper - right ?
I do not support Klipper so far because it is already connected to powerful MCU that can have network

ilbossdelfaidate, 2024-10-21:
I meant to add the accelerometer to the esp8266/esp32 pins to make it calculate the resonant frequency for marlin only on request

luc-github, 2024-10-21:
I did not know Marlin can use Input Shapping based on accelerometer

looking at : https://github.com/Seeed-Studio/Accelerometer_ADXL345
It use Wire and current Lua implementation does not it - I did not expect to go so far 
Which means need to be part of main code of ESP3D 

I do not have any device neither clear user case so PR are welcome

ilbossdelfaidate, 2024-10-21:
in marlin you can use the input shaping by inserting the resonant frequency values ​​and the dumping ratio directly from the menu, I meant if it is possible to create a plugin for esp 3d connected to adxl345 that calculates these values ​​independently from marlin and then inserts them either via M codes or the user manually inserts the values ​​into the printer

luc-github, 2024-10-21:
esp3d does not have pluggin feature only lua scripting feature

luc-github, 2024-11-25:
close due to lack of feedback

closed 2024-11-25 by luc-github