luc-github/ESP3D #1059: [Question]How to Send G-code Commands via HTTP API in ESP3D Version 2.x

userosos, 2024-10-30:
Hello,
I’m using ESP3D version 2.x and want to send G-code commands like G28 Z and M114 through the HTTP API. I've tried requests in the format http://<IP>/command?cmd=M114, but I keep getting an Invalid command error. Could you clarify the correct way to format these requests for version 2.x? Also, if a password is needed, how do I set and include it in the requests?

Thank you for your help!

luc-github, 2024-10-30:
Sorry seems there was a typo in https://esp3d.io/ESP3D/Version_2.1.X/documentation/api/
it is not `cmd=` but `commandText=`

userosos, 2024-10-30:
> Sorry seems there was a typo in https://esp3d.io/ESP3D/Version_2.1.X/documentation/api/ it is not `cmd=` but `commandText=`

Work, thx! If i send g-code - M114 (get current coordinate) - it return only `ok` bit i want get coordinate like in gcode. For example X0 Y0 Z0. Can i do it ftom http request?

luc-github, 2024-10-30:
http command does only [ESPXXX]Responses not others commands

Printer response go to Web socket

if you want same chanel for query/answer you better use the raw tcp whci has default port 8888

userosos, 2024-10-30:
> if you want same chanel for query/answer you better use the raw tcp whci has default port 8888

Hmm. How i can use tcp 8888 port? I create connect from node-red (tcp request node) and it has error and i can see answer. I inject to the node M114.

luc-github, 2024-10-30:
that is depending of you language and API - it is just tcp socket

userosos, 2024-10-30:
> that is depending of you language and API - it is just tcp socket

Hm. I can`t connect to the flow. I can see disconnect.

luc-github, 2024-10-30:
be sure also 8888 is right port - check your settings in webui

userosos, 2024-10-31:
> be sure also 8888 is right port - check your settings in webui

Where i can see it in web UI? In sources i can see port 8888.

luc-github, 2024-10-31:
in settings and in status
https://esp3d.io/ESP3D-WebUI/Version_2.X/showcase/

userosos, 2024-10-31:
> in settings and in status https://esp3d.io/ESP3D-WebUI/Version_2.X/showcase/
I has nosee the port:
[image]

userosos, 2024-10-31:
> well this is because you are not using ESP3D, you are using ESP3DLib for Marlin so that a big difference https://esp3d.io/ESP3DLib/Version_1.X/features/
> 
> ESP3DLib V1.0 does not have Data port (Telnet) support So in that case you must use WebSocket to get Marlin responses like done in WebUI

Oh. Thx! Can i upgrade the board? I use https://github.com/makerbase-mks/MKS-TinyBee

luc-github, 2024-10-31:
yes sure

closed 2024-11-15 by luc-github