luc-github/ESP3D #1066: Online configurator does not specify the pioarduino platform

probonopd, 2024-12-18:
**Describe the bug**

Generate configuration with the online configurator. It does not specify the pioarduino platform.

**Expected behavior**

> Platformio stopped support with espressif for esp32 since 3.0 core update, a community project [pioarduino](https://github.com/pioarduino/platform-espressif32) is now used for esp32 instead, the tests are done using arduino esp32 core 3.0.4 / IDF 5.1.4.

So the online generated configuration should contain

```
[env:esp32s2_4MB_wifi]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
```

but it does not.

**ESP3D Firmware:**

- ESP3D FW version: 3.0

**Board used:**

- MCU: ESP32S2
- Name: S2 Mini
- Flash size: [e.g. 4M: 2M/2M] tbd

luc-github, 2024-12-18:
indeed it was using platformio platform which is stuck to idf 4.4.7 - that is weird - I thought I did it but I do not see any commit...orz

Thank you for reporting it , it is fixed now

closed 2024-12-18 by luc-github

probonopd, 2024-12-19:
Thanks. I tried to compile without changing the config, but it failed. When I changed it manually, it worked. So thanks for having fixed it.