luc-github/ESP3D #1072: [BUG]

CybrGost69, 2025-01-28:
While compiling for Defualt Environment: ESP8266 everything compiles with no errors, when changed to ESP32dev, compile errors develope: 

"Archiving .pioenvs\esp32dev\libb3c\libESPmDNS.a
Archiving .pioenvs\esp32dev\libFrameworkArduino.a
Linking .pioenvs\esp32dev\firmware.elf
.pioenvs\esp32dev\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
.pioenvs\esp32dev\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
.pioenvs\esp32dev\libFrameworkArduino.a(main.cpp.o): In function `loopTask(void*)':
C:\Users\koger\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:14: undefined reference to `setup()'
C:\Users\koger\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:17: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp32dev\firmware.elf] Error 1
======================================================== [FAILED] Took 9.57 seconds ========================================================"

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:09.567

**To Reproduce**

1. Change default_envs from ESP8266 to ESP32dev 
2. Compile
4. See errors

**Expected behavior**

**ESP3D Firmware:**

ESP3D-2.1.x

**Target Firmware:**

N/A

**Board used (please complete the following information):**

ESP32dev

luc-github, 2025-01-29:
The compilation is ok on my side and also on CI 
[image]

I did direct compilation of esp32dev - no need to set as default https://esp3d.io/ESP3D/Version_2.1.X/installation/vscode-platformio/

I have also did same as you build default (esp8266=>Ok), then change default to esp32dev then build default and compilation also passed

[image]

The error you raised is a platform error, not a firmware issue  - it looks like the esp32 sdk is not yet ready on your system 
I suggest you to do a Clean All and redo the compilation

CybrGost69, 2025-02-02:
Roger, that.

closed 2025-02-05 by luc-github