luc-github/ESP3D #1073: [BUG] Uploading is successful but file is 0 bytes

manosheriff, 2025-02-07:
i have Esp3d hooked up to chitu v5 board's UART1 i first slice the file on cura default settings and then it slices the file as gcode file, i am aware of the 8.3 filesystem so i rename the file to be *****.GCO and the file extension is converted the i open the webui and i try to upload this file it starts uploading with no issues and then the file is uploaded to the SD Cardwith only the name with an empty file with 0 bytes and no content and when i try to print this file nothing happens

**ESP3D Firmware:**

- ESP3D FW version: [FW: 1.0]
- ESP3D-WebUI Version:[UI: 2.0c31]
- Flash method: [Arduino Ide]  
- Name: [Marlin]
- Version [02010200]
**Board used**: Chitu v5.1

- MCU:[ESP8266 small Wifi module]
- Flash size: [1M]

[image]

[image]

[image]

[image]

luc-github, 2025-02-08:
these versions are 7 years old, looking a the screenshot I see `Ok P15 NaN GB`  in SD files which means some communication issues and noise during transmission, and also some automatic request send by something during transfert.
If you really want to try a fix I suggest you to move to 2.1 for webUI and esp3d, that the latest version that support the Serial Transfer protocol.

manosheriff, 2025-02-08:
thank you so much i am really big fan of your work it worked thank you for replying

closed 2025-02-08 by manosheriff