luc-github/ESP3D #1074: [BUG-FIX]  The problem with the SSD1306wire library and its solution (add to the main code)

1alexby, 2025-03-04:
**Describe the bug**

The problem is with compiling the code for the esp32-c3 with the enabled oled ssd1306 wire feature in the PlatformIO.

I was able to fix the code myself, and therefore I will attach a file with the corrected .cpp code. The rest can be taken from the original library.

`
#include <Wire.h>

#include "SSD1306wire.h"

typedef void prog_void;
typedef char prog_char;
typedef unsigned char prog_uchar;
typedef char prog_int8_t;
typedef unsigned char prog_uint8_t;
typedef short prog_int16_t;
typedef unsigned short prog_uint16_t;
typedef long prog_int32_t;
typedef unsigned long prog_uint32_t;

#define PROGMEM
#define PGM_P        const char *
#define PGM_VOID_P   const void *
#define PSTR(s)      (s)
#define _SFR_BYTE(n) (n)

#define pgm_read_byte(addr) (*(const unsigned char *)(addr))
#define pgm_read_word(addr)           \
  ({                                  \
    typeof(addr) _addr = (addr);      \
    *(const unsigned short *)(_addr); \
  })
#define pgm_read_dword(addr)         \
  ({                                 \
    typeof(addr) _addr = (addr);     \
    *(const unsigned long *)(_addr); \
  })
#define pgm_read_float(addr)     \
  ({                             \
    typeof(addr) _addr = (addr); \
    *(const float *)(_addr);     \
  })
#define pgm_read_ptr(addr)       \
  ({                             \
    typeof(addr) _addr = (addr); \
    *(void *const *)(_addr);     \
  })

#define pgm_get_far_address(x) ((uint32_t)(&(x)))

#define pgm_read_byte_near(addr)  pgm_read_byte(addr)
#define pgm_read_word_near(addr)  pgm_read_word(addr)
#define pgm_read_dword_near(addr) pgm_read_dword(addr)
#define pgm_read_float_near(addr) pgm_read_float(addr)
#define pgm_read_ptr_near(addr)   pgm_read_ptr(addr)
#define pgm_read_byte_far(addr)   pgm_read_byte(addr)
#define pgm_read_word_far(addr)   pgm_read_word(addr)
#define pgm_read_dword_far(addr)  pgm_read_dword(addr)
#define pgm_read_float_far(addr)  pgm_read_float(addr)
#define pgm_read_ptr_far(addr)    pgm_read_ptr(addr)

#define memcmp_P      memcmp
#define memccpy_P     memccpy
#define memmem_P      memmem
#define memcpy_P      memcpy
#define strcpy_P      strcpy
#define strncpy_P     strncpy
#define strcat_P      strcat
#define strncat_P     strncat
#define strcmp_P      strcmp
#define strncmp_P     strncmp
#define strcasecmp_P  strcasecmp
#define strncasecmp_P strncasecmp
#define strlen_P      strlen
#define strnlen_P     strnlen
#define strstr_P      strstr
#define printf_P      printf
#define sprintf_P     sprintf
#define snprintf_P    snprintf
#define vsnprintf_P   vsnprintf

void SSD1306wire::begin(const SSD1306wire_display* display, uint8_t i2c_addr) {
    _i2c_addr = i2c_addr;
    Wire.begin();
    _max_columns = pgm_read_byte(&display->width);
    _max_rows = pgm_read_byte(&display->height) >> 3;
    const uint8_t* init_sequence = (const uint8_t*)pgm_read_word(&display->init_sequence);
    uint8_t init_sequence_size = pgm_read_byte(&display->init_sequence_size);
    Wire.beginTransmission(_i2c_addr);
    Wire.write(SSD1306_COMMAND);
    for(uint8_t i = 0; i < init_sequence_size; i++) {
        WireSend(SSD1306_COMMAND, pgm_read_byte(init_sequence + i));
    }
    Wire.endTransmission();
    setCursor(0, 0);
}

void SSD1306wire::set400kHz() {
	Wire.setClock(400000);
}

size_t SSD1306wire::write(uint8_t character) {
	if((_column >= _max_columns) || (_row >= _max_rows))
		return 1;
	if(_font == 0)
		return 1;

	uint8_t font_min_width = pgm_read_byte(_font);
	uint8_t font_rows = pgm_read_byte(_font + 1);
	uint8_t font_first_char = pgm_read_byte(_font + 2);
	uint8_t font_last_char = pgm_read_byte(_font + 3);

	uint8_t char_width = 0;
	const uint8_t *char_offset = _font + 5 + font_last_char - font_first_char;
	if((character >= font_first_char) && (character <= font_last_char)) {
		char_width = pgm_read_byte(_font + 4 + character - font_first_char) / font_rows;
		for(uint8_t i = character - font_first_char; i > 0; --i) {
			char_offset += pgm_read_byte(_font + 3 + i);
		}
	}

	uint8_t max_rows = _row + font_rows;
	if (max_rows > _max_rows)
		max_rows = _max_rows;

	for(uint8_t row = _row; row < max_rows; ++row) {
		SetColumn(_column);
		SetRow(row);

		uint8_t c;
		uint8_t i = 0;
		Wire.beginTransmission(_i2c_addr);
		Wire.write(SSD1306_DATA);
		while(i < char_width) {
			if (i + _column >= _max_columns)
				break;
			c = pgm_read_byte(char_offset + i);
			if(_inv_font) c = ~ c;
			Wire.write(c);
			++i;
		}
		c = 0;
		if(_inv_font) c = ~ c;
		while(i < font_min_width) {
			if (i + _column >= _max_columns)
				break;
			Wire.write(c);
			++i;
		}
		Wire.endTransmission();

		char_offset += char_width;
	}

	if(char_width < font_min_width)
		char_width = font_min_width;
	_column += char_width;
	if(_column > _max_columns)
		_column = _max_columns;

	return 1;
}

size_t SSD1306wire::write(const uint8_t *buffer, size_t size) {
	size_t outchars = 0;

	for(int c = 0; c < size; c++) {
		outchars += write(*(buffer + c));
	}

	return outchars;
}

void SSD1306wire::syncCursor() {
	if(_column < _max_columns)
		SetColumn(_column);
	if(_row < _max_rows)
		SetRow(_row);
}

void SSD1306wire::setPower(bool on) {
	Wire.beginTransmission(_i2c_addr);
	Wire.write(SSD1306_COMMAND);
	Wire.write(on ? 0xAF : 0xAE);
	Wire.endTransmission();
}

void SSD1306wire::setContrast(uint8_t contrast) {
	Wire.beginTransmission(_i2c_addr);
	Wire.write(SSD1306_COMMAND);
	Wire.write(0x81);
	Wire.write(contrast);
	Wire.endTransmission();
}

void SSD1306wire::inverseDisplay(bool inverse) {
	Wire.beginTransmission(_i2c_addr);
	Wire.write(SSD1306_COMMAND);
	Wire.write(inverse ? 0xA7 : 0xA6);
	Wire.endTransmission();
}

void SSD1306wire::SetColumn(uint8_t column) {
	Wire.beginTransmission(_i2c_addr);
	Wire.write(SSD1306_COMMAND);
	Wire.write(0x00 | (column & 0x0F));
	Wire.write(0x10 | (column >> 4));
	Wire.endTransmission();
}

void SSD1306wire::SetRow(uint8_t row) {
	Wire.beginTransmission(_i2c_addr);
	Wire.write(SSD1306_COMMAND);
	Wire.write(0xB0 | row);
	Wire.endTransmission();
}

void SSD1306wire::clear() {
	SetColumn(0);

	for(uint8_t row = 0; row < _max_rows; ++row) {
		SetRow(row);

		Wire.beginTransmission(_i2c_addr);
		Wire.write(SSD1306_DATA);
		for(uint8_t col = 0; col < _max_columns; ++col) {
			WireSend(SSD1306_DATA, 0);
		}
		Wire.endTransmission();
	}

	setCursorForce(0, 0);
}

void SSD1306wire::clearToEOL() {
	if((_column >= _max_columns) || (_row >= _max_rows))
		return;
	if(_font == 0)
		return;

	uint8_t max_rows = _row + pgm_read_byte(_font + 1);
	if (max_rows > _max_rows)
		max_rows = _max_rows;

	uint8_t c = 0;
	if(_inv_font) c = ~ c;

	for(uint8_t row = _row; row < max_rows; ++row) {
		SetColumn(_column);
		SetRow(row);

		Wire.beginTransmission(_i2c_addr);
		Wire.write(SSD1306_DATA);
		for(uint8_t col = _column; col < _max_columns; ++col) {
			WireSend(SSD1306_DATA, c);
		}
		Wire.endTransmission();
	}
	_column = _max_columns;
}

void SSD1306wire::WireSend(uint8_t type, uint8_t value) {
	if(Wire.write(value) == 0) {
		Wire.endTransmission();
		Wire.beginTransmission(_i2c_addr);
		Wire.write(type);
		Wire.write(value);
	}
}`

Ahem, I'm sorry, the code can't be inserted otherwise........

luc-github, 2025-03-04:
whet esp3d version are you refering to
please share your configuration file and plarformio.ini to  reproduce isdue

1alexby, 2025-03-09:
> на какую версию esp3d вы ссылаетесь? пожалуйста, поделитесь файлом конфигурации и plarformio.ini, чтобы воспроизвести проблему

I'm referring to version 3.0.

[platformio.txt](https://github.com/user-attachments/files/19152720/platformio.txt)

luc-github, 2025-03-10:
I used your platformio file and cannot see any issue , you did not described the issue and the changes you did to solve the issue, you did not provided the configuration.h as I requested so I have no way to investigate sorry

[image]

luc-github, 2025-03-21:
@1alexby any update ?

1alexby, 2025-03-21:
> @1alexby any update ?

No, I'm taking a break on this topic for now, I'll try again when I have time.

The main problem was that in platform io the code did not compile because there were errors related to the library ssd1306 displays. But perhaps the problem arose for some reason only with me. I had not used it before platform io and it was clean, I took the whole configuration from web constructor, so there couldn't be any mistakes on my part. Again, I used the most common esp32-c3 supermini. Problems arose already at the compilation stage, I made a small fix which I posted here.

luc-github, 2025-03-21:
well I cannot duplicate issue so I close it

closed 2025-03-21 by luc-github