luc-github/ESP3D #1075: [BUG] The image from the camera freezes (AI Thinker ESP32-CAM)

vladikas, 2025-03-06:
**Describe the bug**

The video camera shows one frame and freezes.

**To Reproduce**

1. Go to .../esp32cam.html
2. Click on "Start stream"
3. Get one frame
4. See error - the image is frozen

**Expected behavior**

When installing the full firmware [ESP32-CAM_MJPEG2SD](https://github.com/s60sc/ESP32-CAM_MJPEG2SD)  on the same board  everything works fine.

**ESP3D Firmware:**

- ESP3D FW version: 3.0 
- ESP3D-WebUI Version: 3.0 
- Wifi mode: any
- Flash method: Arduino IDE 2.3.4
    If you build with Arduino:
    - Arduino ESP32 Espressif package version: 3.1.3
- Any change done in FW:
`
#ifndef _CONFIGURATION_H
#define _CONFIGURATION_H

#if defined __has_include
#if __has_include("myconfig.h")
#include "myconfig.h"
#endif
#endif

#define ESP32_WIFI_TX_POWER WIFI_POWER_15dBm

#define COMMUNICATION_PROTOCOL RAW_SERIAL

#define ESP_SERIAL_OUTPUT USE_SERIAL_0

#define SERIAL_RX_BUFFER_SIZE 1024

#define DEFAULT_FW MARLIN

#define WIFI_FEATURE

#define ESP3D_ETH_PHY_TYPE TYPE_ETH_PHY_W5500

#define ETHERNET_SPI_USE_SPI 1
#define ETHERNET_SPI_USE_SPI2 0
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19
#define ETH_PHY_CS 5
#define ETH_PHY_IRQ -1
#define ETH_PHY_RST 4

#define ESP3D_ETH_PHY_ADDR 1

#define HTTP_FEATURE

#define DISABLE_TELNET_WELCOME_MESSAGE

#define NOTIFICATION_FEATURE

#define NOTIFICATION_ESP_ONLINE "Hi, %ESP_NAME% is now online at %ESP_IP%"

#define ESP_NOTIFICATION_TITLE "ESP3D Notification"

#define CAPTIVE_PORTAL_FEATURE

#define MDNS_FEATURE

#define SSDP_FEATURE

#define ESP_MODEL_NAME "ESP Board"

#define ESP_MODEL_NUMBER "ESP3D 3.0"

#define ESP_MODEL_URL "https://www.espressif.com/en/products/devkits"

#define ESP_MANUFACTURER_NAME "Espressif Systems"

#define ESP_MANUFACTURER_URL "https://www.espressif.com"

#define FILESYSTEM_FEATURE ESP_LITTLEFS_FILESYSTEM

#define FILESYSTEM_TIMESTAMP_FEATURE

#define SD_DEVICE_CONNECTION ESP_NOT_SHARED_SD

#define SD_DEVICE ESP_SDIO

#define SDIO_BIT_MODE SD_FOUR_BIT_MODE

#define SD_TIMESTAMP_FEATURE

#define GLOBAL_FILESYSTEM_FEATURE

#define FTP_FEATURE FS_ROOT

#define WEB_UPDATE_FEATURE

#define SD_UPDATE_FEATURE

#define PRINTER_HAS_DISPLAY

 #define DISPLAY_I2C_ADDR 0x3c
 #define ESP_SDA_PIN 21
 #define ESP_SCL_PIN 22

#define SENSOR_ADDR 0x76

#define SENSOR__UNIT "C"

#define CAMERA_DEVICE CAMERA_MODEL_AI_THINKER

#define SERIAL_COMMAND_FEATURE

#define DIRECT_PIN_FEATURE

#define ESP_AUTOSTART_SCRIPT_FILE "autoscript.gco"

#define ESP_GOT_IP_HOOK "[ESP212]IP:%ESP_IP%"

#define ESP_GOT_DATE_TIME_HOOK "[ESP212]DATE:%ESP_DATETIME%"

#define GCODE_HOST_FEATURE

#define ESP_SAVE_SETTINGS SETTINGS_IN_EEPROM

#ifdef ESP_LOG_FEATURE
#define LOG_ESP3D_BAUDRATE 115200
#define LOG_ESP3D_OUTPUT_PORT 8000
#endif 

#if defined(ESP_GOT_DATE_TIME_HOOK) || defined(SD_TIMESTAMP_FEATURE) || \
    defined(FILESYSTEM_TIMESTAMP_FEATURE)
#define TIMESTAMP_FEATURE
#endif  // SD_TIMESTAMP_FEATURE || FILESYSTEM_TIMESTAMP_FEATURE

#if defined(PRINTER_HAS_DISPLAY)
#define HAS_SERIAL_DISPLAY ""
#endif  // PRINTER_HAS_DISPLAY

#if defined(CAMERA_DEVICE)
#if CAMERA_DEVICE == CAMERA_MODEL_ESP32_CAM_BOARD || \
    CAMERA_DEVICE == CAMERA_MODEL_ESP32S2_CAM_BOARD
#define USE_BOARD_HEARDER 1
#endif  // CAMERA_DEVICE==CAMERA_MODEL_ESP32_CAM_BOARD ||
        // CAMERA_DEVICE==CAMERA_MODEL_ESP32S2_CAM_BOARD
#endif  // CAMERA_DEVICE

#if !defined(WIFI_FEATURE) && !defined(ETH_FEATURE)
#undef HTTP_FEATURE
#undef TELNET_FEATURE
#undef WEBDAV_FEATURE
#undef FTP_FEATURE
#undef WEB_UPDATE_FEATURE
#undef CAPTIVE_PORTAL_FEATURE
#undef SSDP_FEATURE
#undef MDNS_FEATURE
#undef NOTIFICATION_FEATURE
#endif

#endif  //_CONFIGURATION_H`
**Target Firmware:**

- Name: Marlin
- Version  2.1

**Board used (please complete the following information):**

- MCU: ESP32
- Name: AI Thinker ESP32-CAM
- Flash size: 4M: Huge App (3MB No OTA/1MB SPIFFS)

**Additional context**
Verbose log
[CoolTerm Capture (Untitled_0) 2025-03-06 09-51-57-430.txt](https://github.com/user-attachments/files/19103835/CoolTerm.Capture.Untitled_0.2025-03-06.09-51-57-430.txt)

vladikas, 2025-03-06:
Everything works. I'm sorry to have taken up your time.

closed 2025-03-06 by vladikas

luc-github, 2025-03-06:
good have fun, np