luc-github/ESP3D #1090: [Question] FYSETC SD-WIFI-PRO compile with SD support

spleijers, 2025-04-04:
Hello, im trying to flash a FYSETC SD-WIFI-PRO with ESP3D (v3) with the configuration.h/ ini provided [here](https://esp3d.io/ESP3D/Version_3.X/hardware/esp_boards/esp32-pico/sd-wifi-pro/); 

While compiling i ran into an issue with some usb related library, so after some tries i deleted the folder /libraries/esp32-usb-serial-1.0.1 and suddenly flashing worked!

This was the error;
```Compiling .pioenvs\esp32_4MB_wifi\lib144\esp32-usb-serial-1.0.1\usb_host_vcp.cpp.o
In file included from libraries/esp32-usb-serial-1.0.1/src/esp32_usb_serial.h:22,
                 from libraries/esp32-usb-serial-1.0.1/src/esp32_usb_serial.cpp:24:
libraries/esp32-usb-serial-1.0.1/src/usb/cdc_acm_host.h:10:10: fatal error: usb/usb_host.h: No such file or directory
   10 | #include "usb/usb_host.h"
      |          ^~~~~~~~~~~~~~~~
```
Now the only problem that remains; SD card access does not seem to function;

[image]

Is that because i removed that library, or is there something else that needs to be changed?

luc-github, 2025-04-04:
What version of  ESP3D are you using ? please check the webui information page to give the exact version

spleijers, 2025-04-04:
Now thats a quick reply! :) This is the version;

[image]

luc-github, 2025-04-04:
about usb-host issue the platformio.ini miss the line 
`lib_ignore = esp32-usb-serial` I will update the doc which seems a little bit outdated

About SD issue I cannot duplicate, be sure your SD speed is same as screenshot :

[image]

spleijers, 2025-04-04:
Ah yes, then deleting the /libraries/esp32-usb-serial-1.0.1 folder would have had the same effect. For now i added that line to platformio.ini. 

I checked the SD card config and it is set to 4 and SD updater 'Yes'. As for the SD card itself, should the 'slider' on the side be switched to the top?

I also tried formatting the SD card with the [SD card formatter](https://www.sdcard.org/downloads/formatter/) but nothing changed.

spleijers, 2025-04-04:
Perhaps there is a change needed here?; 

[image]

luc-github, 2025-04-04:
Yes SD need to be enabled  and save setting when it is orange

spleijers, 2025-04-04:
Yes, unfortunately it does not seem to work when switched on. Maybe some other settings in the SD play commands? Or other filesystem?

luc-github, 2025-04-04:
does it work when you are connected with your daughter board ?

[image]

[image]

spleijers, 2025-04-04:
Wow, yes! Good catch, that seems to be working!

Strange, if i plug it in a windows machine with builtin cardreader it wont work at all... Seems to be 'locking' the SD card?

luc-github, 2025-04-04:
yes windows lock the SD card, or try eject menu in windows  SDCard menu

closed 2025-05-05 by luc-github