luc-github/ESP3D #1092: [Question]use of readData() on lua interpreter

kleffa, 2025-05-05:
local pin = readData([ESP201] P=4)
if pin == 1 then
    print("[ESP201] P=4 V=0 ")
else
    print("[ESP201] P=4 V=1 ")
end

I'm writing a couple of scripts like this one (just change state of flash on esp32cam). I know this can be just a digitalRead, but its about using readData() , this doesnt work, I missed something?
can readData used to read serial, like M27 data?

any help please to clarify it

luc-github, 2025-05-05:
readData() read any string received by lua interpreter, from ESP3D if you send any [ESPXXX] command but also any string from serial  or usb  
use available() to check if any string is available then readData() to collect one string at once

here a sample script that echo any message received by lua interpreter
```
--[[
    Echo Serial

    This example echo serial outout to serial with echo: header 
    Put this script in the ESP3D flash file system and call it from the init script in 
    configuration.h.
    #define ESP_AUTOSTART_SCRIPT "[ESP300]/FS/detectpush.lua"

    Be sure to have the following line comment out in the configure.h:
    #define ESP_LUA_INTERPRETER_FEATURE
    ]]--

-- Setup
local availableMsg = 0
local message = ""
while (true) do
    -- check if have data
    availableMsg = available()
    -- if any message
    if (availableMsg > 0) then
        message = "echo:" .. readData()
        print(message)
    end
    -- yield to other tasks
    yield()
end
```

I know documentation https://esp3d.io/ESP3D/Version_3.X/documentation/lua/ is very light but no one seems used this feature until now because I got no feedback on this feature, so feel free to propose, submit better documentation, of course I will help by answering any question or give clarifications if necessary

luc-github, 2025-05-06:
@kleffa did this answered your question ?

kleffa, 2025-05-06:
Yes, let me try some scripts and share as example

kleffa, 2025-06-26:
--[[
    FLASH BUTTON

    This example is implemented in a ESP32-CAM.

    in CURA post procesing tab, find time lapse, there is the option to set parking hotend

    You can set a macros, as a GCODE one, with this command
    [ESP300]/FS/flash.lua
    
    You can also start it manually from the serial monitor by typing:
    [ESP300]/FS/flash.lua
    
    Be sure to have the following line comment out in the configure.h:
    #define ESP_LUA_INTERPRETER_FEATURE
    ]]--

local FLASH_PIN = 4          -- GPIO for flash

pinMode(FLASH_PIN, OUTPUT) 
local STATE = digitalRead(FLASH_PIN)

if STATE == HIGH then
    digitalWrite(FLASH_PIN, LOW)
else
    digitalWrite(FLASH_PIN, HIGH)
end

kleffa, 2025-06-26:
> great thank you, I am happy lua is helpful for something, I did not get so
> many feedback

I think there should be a way to convine M118 and lua to make a variable to
save the timelapse pics on sd/{jobname}/

Any idea is welcome

luc-github, 2025-07-29:
I close topic as answered and no new activity

closed 2025-07-29 by luc-github