luc-github/ESP3D #1098: [BUG]<Enter comprehensive title>

fedekrum, 2025-09-29:
**Describe the bug**

Hi. While compiling, I get this:

/Users/fede/Downloads/ESP3D-3.0/esp3d/src/core/commands/ESP420.cpp: In member function 'void ESP3DCommands::ESP420(int, ESP3DMessage*)':
/Users/fede/Downloads/ESP3D-3.0/esp3d/src/core/commands/ESP420.cpp:316:17: error: 'class SSDPClass' has no member named 'started'
  316 |   tmpstr = SSDP.started() ? "ON (" + String(SSDP.localIP().toString()) + ")" : "OFF";
      |                 ^~~~~~~
/Users/fede/Downloads/ESP3D-3.0/esp3d/src/core/commands/ESP420.cpp:316:57: error: 'IPAddress SSDPClass::localIP()' is protected within this context
  316 |   tmpstr = SSDP.started() ? "ON (" + String(SSDP.localIP().toString()) + ")" : "OFF";
      |                                             ~~~~~~~~~~~~^~
In file included from /Users/fede/Downloads/ESP3D-3.0/esp3d/src/core/commands/ESP420.cpp:85:
/Volumes/Development/Arduino/libraries/ESP32SSDP/src/ESP32SSDP.h:117:13: note: declared protected here
  117 |   IPAddress localIP();
      |             ^~~~~~~
exit status 1

Compilation error: 'class SSDPClass' has no member named 'started'

**To Reproduce**

Compile

**Expected behavior**
to Compile

**ESP3D Firmware:**
3.0.0

**Target Firmware:**

Marlin 2.0

**Board used (please complete the following information):**

ESP32 S3 Dev Board - 16MB Flash, 8MB PSRAM

**Additional context**

luc-github, 2025-09-29:
so you did not used the one provided....

fedekrum, 2025-09-29:
inside of libraries in the 3.0.0 version, I have ESP32SSDP-2.0.3

luc-github, 2025-09-29:
and this is is the one you copied to arduino library directory ?

fedekrum, 2025-09-29:
My fault !!! I copied all the libs to an old libraries directory not used any more by my Arduino IDE program. 
Sorry !!! And Thanks